mdegans/nano_build_opencv #4: Error when building CXX object modules/python2/CMakeFiles/opencv_python2.dir/__/src2/cv2.cpp.o

monocongo, 2019-05-17:
Thanks for providing this script, as I will need OpenCV for software I plan to run on my Jetson Nano board.

I have run the build script and it failed right near the end. [[please see the output copied from the console](https://github.com/mdegans/nano_build_opencv/files/3191823/opencv_build_output.txt)]

This was run on a fresh Jetson Nano. The only thing I've done there before running the build script was to update/install a few packages (following this [tutorial](https://www.pyimagesearch.com/2019/05/06/getting-started-with-the-nvidia-jetson-nano/)):

```
$ sudo apt-get update
$ sudo apt-get upgrade
$ sudo apt-get install git cmake
$ sudo apt-get install libatlas-base-dev gfortran
$ sudo apt-get install libhdf5-serial-dev hdf5-tools
$ sudo apt-get install python3-dev
$ cd ~/git 
$ git clone https://github.com/mdegans/nano_build_opencv.git
$ cd nano_build_opencv
$ ./build_opencv.sh 
```

The final message in the output (where the fatal error occurs?) appears to be Python2 related. Is it necessary to build OpenCV for Python2 support? I don't plan to use Python2 (all my Python code is Python 3.6+), so maybe I can eliminate the parts that are building for Python2 support and maybe bypass this error as a result?

I apologize I'm unfamiliar with OpenCV builds, compilation error messages, etc. so I'm dependent on others to make sense of this for me. If I can help debug or troubleshoot this issue further from my end then please ask so I can help you help me.

mdegans, 2019-05-20:
I'd be happy to help. It looks like it's failing at the building python2 module. One possible troubleshooting step is to try to build without python2 and see if that fixes it. [here](https://github.com/mdegans/nano_build_opencv/blob/no-python2/build_opencv.sh) is a version with no Python2 support. Please let me know if it builds.

Also, please note that building OpenCV requires a lot of memory. If you encounter failures at random points, you might consider running headless (ssh in and then `systemctl isolate multi-user.target`). A swap file might also help, but I would recommend mounting it on an external drive if you do use one. Swapping from a MicroSD card severely hurts it's lifespan.

monocongo, 2019-05-21:
Thanks, @mdegans. I will give this a try.

I appreciate the heads up about running a swap file on my microSD card, as I just set that up today. I guess I will switch over to an external hard drive for swap.

I was able to get OpenCV version 4.0 to build using this [script from NVIDIA](https://github.com/AastaNV/JEP/blob/master/script/install_opencv4.0.0_Nano.sh).

mdegans, 2019-05-21:
> Thanks, @mdegans. I will give this a try.
> 
> I appreciate the heads up about running a swap file on my microSD card, as I just set that up today. I guess I will switch over to an external hard drive for swap.

That's a good solution, just make sure it's usb3 and fast. You can use the `hdparm` utility to benchmark drives on linux. For example `sudo hdparm -t /dev/mmcblk0` will read benchmark your MicroSD on linux. If it's not installed, you can install it with `sudo apt-get install hdparm`. I have benchmarked external SSDs at around 250MB/s so that's the number you should be expecting from `/dev/sda` (probably the name of your first usb drive).

> I was able to get OpenCV version 4.0 to build/install using this [script from NVIDIA](https://github.com/AastaNV/JEP/blob/master/script/install_opencv4.0.0_Nano.sh).

That script works perfectly well, but please be aware that you will likely have similar issues if you try to build without enough free memory. It will probably work fine until it comes time to link everything together at the end, and then fail at some random thing, yielding no useful google results from the error message.

m7mdosama, 2019-11-17:
> I'd be happy to help. It looks like it's failing at the building python2 module. One possible troubleshooting step is to try to build without python2 and see if that fixes it. [...]

The link is not found , Please update

DaveXanatos, 2019-11-28:
Please update link for no-python2/build, as I'm encountering this same issue.

Thanks!

mdegans, 2020-03-03:
> Please update link for no-python2/build, as I'm encountering this same issue.
> 
> Thanks!

Sorry for the necropost. I removed that becuase it wasn't the root of the problem. If you are having the same issue, please try building again with a swap file / partition.

closed 2020-03-18 by mdegans