mdegans/nano_build_opencv #10: fatal error: Eigen/Core: No such file or directory

Hongbo-Miao, 2019-07-31:
Got this error when try to run the script on Jetson Nano.

```
...
[ 25%] Built target opencv_cudev
Scanning dependencies of target opencv_core_pch_dephelp
[ 25%] Building CXX object modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/opencv_core_pch_dephelp.cxx.o
In file included from /tmp/build_opencv/opencv/modules/core/src/precomp.hpp:55:0,
                 from /tmp/build_opencv/opencv/build/modules/core/opencv_core_pch_dephelp.cxx:1:
/tmp/build_opencv/opencv/modules/core/include/opencv2/core/private.hpp:66:12: fatal error: Eigen/Core: No such file or directory
 #  include <Eigen/Core>
            ^~~~~~~~~~~~
compilation terminated.
modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/build.make:62: recipe for target 'modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/opencv_core_pch_dephelp.cxx.o' failed
make[2]: *** [modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/opencv_core_pch_dephelp.cxx.o] Error 1
CMakeFiles/Makefile2:2699: recipe for target 'modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/all' failed
make[1]: *** [modules/core/CMakeFiles/opencv_core_pch_dephelp.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

mdegans, 2019-08-01:
@Hongbo-Miao 

Please see this issue for solutions:
https://github.com/opencv/opencv/issues/14868

It may actually be fixed already.

closed 2019-08-01 by mdegans

Hongbo-Miao, 2019-08-01:
Hi @mdegans I would say it works perfectly on 4.1.0, however the issue came back on 4.1.1 after I changed your script [this line](https://github.com/mdegans/nano_build_opencv/blob/master/build_opencv.sh#L8) to
```
readonly DEFAULT_VERSION=4.1.1
```

Hongbo-Miao, 2019-08-01:
@mdegans oh, I just saw it got merged yesterday. Thanks! Would like to try again in future on OpenCV 4.1.1 when got chance!