mdegans/nano_build_opencv #9: [[: not found

Hongbo-Miao, 2019-07-31:
Thanks for the script!

Just some errors when I try to run the script, but won't affect the script running.

> build_opencv.sh: 110: build_opencv.sh: [[: not found
> build_opencv.sh: 114: build_opencv.sh: [[: not found
> build_opencv.sh: 33: build_opencv.sh: [[: not found

mdegans, 2019-08-01:
@Hongbo-Miao 

Thank you for reporting the issue. I can't replicate.

- What version of bash are you running  (`bash --version`)?
- What version of the script are you running (it should have `#!/usr/bin/env bash` at the top of the file)?

Hongbo-Miao, 2019-08-01:
1) I am using the macOS to ssh into Jetson Nano to run the script.

This is what returned after running
```
bash --version
```
on my macOS.

> GNU bash, version 3.2.57(1)-release (x86_64-apple-darwin18)
> Copyright (C) 2007 Free Software Foundation, Inc.

2) I downloaded and ran this script two days ago, it has `#!/usr/bin/env bash` on the top of the file.

mdegans, 2019-08-01:
Sorry for not specifying. I meant the version of bash on the nano, not your host.

The version of bash reported by my nano is:
> GNU bash, version 4.4.20(1)-release (**aarch64**-unknown-**linux**-gnu)

When you ssh into the nano run the build script, are you running it directly like `./build_opencv.sh` or are you doing `sh build_opencv.sh`? If it's the latter, that could explain the error. The latter would use the bourne shell (/bin/sh) or similar, rather than bash, and some syntax is slightly different. It for sure won't work with /bin/sh.

Hongbo-Miao, 2019-08-02:
Oh, I think you are right, I remembered I was using `sh build_opencv.sh`. Thanks for explanation!

closed 2019-08-02 by Hongbo-Miao