mdegans/nano_build_opencv #11: c++: internal compiler error: Killed (program cc1plus)

AbhimanyuAryan, 2019-08-02:
I getting this error while building OpenCV. I appears it's using Python2 to build it. Should that be a problem? Or should it use Python3 if yes how & wha to change?

```
      Bases:  cv::Algorithm, cv::class, cv::Feature2D, cv::Algorithm
      Only the first base class will be used
Note: Class detail_GraphCutSeamFinder has more than 1 base class (not supported by Python C extensions)
      Bases:  cv::detail::GraphCutSeamFinderBase, cv::detail::SeamFinder
      Only the first base class will be used
[ 98%] Built target gen_opencv_python_source
Scanning dependencies of target opencv_python2
[ 98%] Building CXX object modules/python2/CMakeFiles/opencv_python2.dir/__/src2/cv2.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
modules/python2/CMakeFiles/opencv_python2.dir/build.make:62: recipe for target 'modules/python2/CMakeFiles/opencv_python2.dir/__/src2/cv2.cpp.o' failed
make[2]: *** [modules/python2/CMakeFiles/opencv_python2.dir/__/src2/cv2.cpp.o] Error 4
CMakeFiles/Makefile2:10668: recipe for target 'modules/python2/CMakeFiles/opencv_python2.dir/all' failed
make[1]: *** [modules/python2/CMakeFiles/opencv_python2.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```

mdegans, 2019-08-02:
Please see issue #4 

Your issue is most often caused by trying to run the build without enough free memory. It will fail at the the end of the build like this if it does. Please try again with a swapfile mounted and/or run headless. To turn off X temporarily (after you ssh into your nano) please run `systemctl isolate multi-user.target`.

closed 2019-08-02 by mdegans

AbhimanyuAryan, 2019-08-06:
@mdegans there was enough free memory

mdegans, 2019-08-09:
If there was enough free memory, I'd suggest submitting a gcc bug report by following the instructions in the error message, if you can replicate it. I cannot. Can you tell me what options you ran the script on and what version of Linux for Tegra you are using (32.2, 32.1, etc).

nerdpoplu, 2024-07-02:
Did you get the solution to that ?? because even I'm facing similar problems 
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
modules/python2/CMakeFiles/opencv_python2.dir/build.make:62: recipe for target 'modules/python2/CMakeFiles/opencv_python2.dir//src2/cv2.cpp.o' failed
make[2]: *** [modules/python2/CMakeFiles/opencv_python2.dir//src2/cv2.cpp.o] Error 4
CMakeFiles/Makefile2:7027: recipe for target 'modules/python2/CMakeFiles/opencv_python2.dir/all' failed
make[1]: *** [modules/python2/CMakeFiles/opencv_python2.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Linking CXX shared module ../../lib/python3/cv2.cpython-36m-aarch64-linux-gnu.so
[100%] Built target opencv_python3
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2