mdegans/nano_build_opencv #12: Can we use same script for OpenCV's older versions like 3.3.1?

archanabwk, 2020-02-10:
I have OpenCV 3.3.1 installed on Nano. I want to  rebuild it with Cuda enabled on it. Can we use same script for OpenCV 3.3.1 or any older version than OpenCV 4?

mdegans, 2020-03-03:
Probably. You can [specify a version like this](https://github.com/mdegans/nano_build_opencv#specifying-an-opencv-version-git-branch). Please let me know if any problems with a specific version and I will try to adjust the script to be more flexible to accomadate.

archanabwk, 2020-03-03:
@mdegans Thanks for the reply. For now I have reinstalled the OpenCV with latest version on Nano and able to build it with Cuda enabled. I will try this version specific re-compilation on another Nano.

a904418, 2020-08-05:
I tried installing opencv_python==3.4.3.18 version and I get this message:

Getting version '3.4.3.18' of OpenCV
Cloning into 'opencv'...
warning: Could not find remote branch 3.4.3.18 to clone.
fatal: Remote branch 3.4.3.18 not found in upstream origin

lweingart, 2021-01-02:
Hi mdegans,

First, I would like to thank you very much for providing this script. It allowed me to install opencv 4.4 flawlessly on my jetson nano jetpack 4.4, so really, thank you!

I'm trying to make an old piece of code to work on my jetson nano as well, and this project uses opencv 3.2.0.
I decided to try your script to install this opencv version but unfortunately it fails, after some time.
The fail seems to be related to cmake configuration, but as I have no knowledge of cmake, I'm struggling to find the reason why.

If by chance you have some free time and you would like to help me use your script to install opencv 3.2.0, [here](https://gist.github.com/lweingart/69d59ed422108f699daf85d8dc0a71e5) is a link to my terminal output, the CMakeOutput.log and the CMakeError.log

Thank you, happy new year!

Laurent