mdegans/nano_build_opencv #13: Did not build for python3

lackdaz, 2020-02-17:
I ran the build script and opencv4 was installed only for Python 2 and not 3 - with python 3 defaulting to '3.3.1'. Any clues what might be the case?

mdegans, 2020-03-03:
It's possible your build ran out of memory in that part of the build. It's usually what causes this kind of issue. Do you have a swap file/partition mounted?

lackdaz, 2020-03-04:
@mdegans 
I do have a swap file resized for 4gb, but perhaps I need more.

I used the jetsonhacks script eventually and that one built for python3, I could try running it again later today when I need to build OpenCV 4.2 for cudnn.

Btw thanks for your contributions on the nvidia devtalks forums, your posts have helped me immeasurably!

mdegans, 2020-03-13:
Glad you find my posting helpful, and thanks for the thanks. You could try with a larger swap file, but it's possible something else is going on.

Are you getting any error messages during the build? What happens when you run `opencv_version --verbose` ?

mdegans, 2020-03-18:
Any updates? If not, I will probably close this as it's similar to other issues related to memory use at build time.

closed 2020-03-18 by mdegans