mdegans/nano_build_opencv #19: OpenCV 4.2.0 build fails at opencv_visualisation

domcross, 2020-03-30:
Trying to build latest OpenCV (4.2.0) on Jetson Nano (Jetpack 4.3 / L4T 32.3.1), but it fails at opencv_visualisation:
```
Scanning dependencies of target opencv_visualisation
[ 75%] Building CXX object apps/visualisation/CMakeFiles/opencv_visualisation.dir/opencv_visualisation.cpp.o
[ 76%] Building CXX object modules/dpm/CMakeFiles/opencv_dpm.dir/src/dpm_model.cpp.o
[ 76%] Linking CXX executable ../../bin/opencv_visualisation
../../lib/libopencv_videoio.so.4.2.0: undefined reference to `av_packet_unref'
../../lib/libopencv_videoio.so.4.2.0: undefined reference to `avformat_get_mov_video_tags'
collect2: error: ld returned 1 exit status
apps/visualisation/CMakeFiles/opencv_visualisation.dir/build.make:100: recipe for target 'bin/opencv_visualisation' failed
make[2]: *** [bin/opencv_visualisation] Error 1
CMakeFiles/Makefile2:6929: recipe for target 'apps/visualisation/CMakeFiles/opencv_visualisation.dir/all' failed
make[1]: *** [apps/visualisation/CMakeFiles/opencv_visualisation.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
```
Thanks in advance.

mdegans, 2020-04-01:
Thanks for the report. I'm trying to replicate now.

mdegans, 2020-04-02:
I ran the build yesterday on Nano an Xavier to test without a problem on the same JetPack. The most common cause of these kind of failures is running out of memory during build. Couple questions:

- Are you running with the GUI enabled?
- Do you have a swapfile mounted?

domcross, 2020-04-03:
Thanks for looking into this. I have GUI disabled and 8GB swapfile mounted. After system startup there is 0.3GB out 4.0GB RAM used and total swap is 10GB. 

I ran `bash build_opencv.sh` immediately after a reboot and still see the same error. 

Note: currently OpenCV 3.2.0 is installed from a source build (not using your script).

mdegans, 2020-04-03:
I was able to complete the builds on Xavier and Nano yesterday with only zram swap and the GUI off. If you see the same error at the same point in the middle of the build it's *probably* not memory related. Most failures i've seen happen at the end with the python bindings.

These errors would seem to be ffmpeg related. Do you have ffmpeg installed, and if so, what versions?

domcross, 2020-04-04:
yes, the error seems to be related to some dependencies...
My ffmpeg version:
```
ffmpeg version 2.0.7 Copyright (c) 2000-2015 the FFmpeg developers
  built on Mar 13 2020 10:05:08 with gcc 7 (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04)
  configuration: --disable-doc
  libavutil      52. 38.100 / 52. 38.100
  libavcodec     55. 18.102 / 55. 18.102
  libavformat    55. 12.100 / 55. 12.100
  libavdevice    55.  3.100 / 55.  3.100
  libavfilter     3. 79.101 /  3. 79.101
  libswscale      2.  3.100 /  2.  3.100
  libswresample   0. 17.102 /  0. 17.102
```

mdegans, 2020-04-07:
So it looks like your version is not the same as mine. I'm not sure how that's happening.
```
 $ ffmpeg --version
ffmpeg version 3.4.6-0ubuntu0.18.04.1 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7 (Ubuntu/Linaro 7.3.0-16ubuntu3)
  configuration: --prefix=/usr --extra-version=0ubuntu0.18.04.1 --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-librsvg --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
  libpostproc    54.  7.100 / 54.  7.100
```
Your seems to be significantly older, and was built using different config flags. That shouldn't happen on JetPack 4.3 with the version in Canonical's repos (ports.ubuntu.com). Did you build ffmpeg yourself, or do you have some custom version installed? There is a custom version of FFMPEG supplied by a third party for Tegra, but it's untested with OpenCV as far as I am aware.

The easiest solution is probably to uninstall whatever version you have or specify `-D WITH_FFMPEG=OFF` as a cmake flag (you can add this around line 121, with the rest of the flags). If you do need ffmpeg support, a version that works* is in the apt repos (`sudo apt install libavcodec-dev`)

*I would advise to use gstreamer for any video decoding at the moment since ffmpeg currently lacks hw decode on Tegra (however Nvidia has announced plans to include it in the next JetPack release).

domcross, 2020-04-09:
I ran the script in a fresh docker container and it finished without any errors. `cv2.__version__` returns "4.2.0".

So issue was my messed up environment (the outdated ffmpeg was a dependency from the OpenCV 3.2 that I built manually). 

Again, thanks for your support and thanks for the script.

closed 2020-04-09 by domcross