mdegans/nano_build_opencv #22: Stuck at Checking for module 'gstreamer-app-1.0'

Sayyam-Jain, 2020-04-13:
Hi, thank you for the shell file
I'm trying to run the shell file inside Dockerfile using l4t-base:r32.2 as base image in a qemu emulator.

While installing, it shows the following output:
```
-- Performing Test HAVE_CXX_WNO_ENUM_COMPARE_SWITCH - Failed
-- Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY JAVA_INCLUDE_PATH JAVA_INCLUDE_PATH2 JAVA_AWT_INCLUDE_PATH) 
-- VTK is not found. Please set -DVTK_DIR in CMake to VTK build directory, or to VTK install subdirectory with VTKConfig.cmake file
-- Performing Test CXX_HAS_MFPU_NEON
-- Performing Test CXX_HAS_MFPU_NEON - Failed
-- Performing Test C_HAS_MFPU_NEON
-- Performing Test C_HAS_MFPU_NEON - Failed
-- Looking for dlerror in dl
-- Looking for dlerror in dl - found
-- Performing Test HAVE_C_WNO_UNDEF
-- Performing Test HAVE_C_WNO_UNDEF - Success
-- ADE: Download: v0.1.1f.zip
qemu: Unsupported syscall: 278
qemu: Unsupported syscall: 278
-- OpenCV Python: during development append to PYTHONPATH: /tmp/build_opencv/opencv/build/python_loader
-- Performing Test HAVE_CXX_WNO_STRICT_ALIASING
-- Performing Test HAVE_CXX_WNO_STRICT_ALIASING - Success
-- Checking for modules 'libavcodec;libavformat;libavutil;libswscale'
--   Found libavcodec, version 57.107.100
--   Found libavformat, version 57.83.100
--   Found libavutil, version 55.78.100
--   Found libswscale, version 4.8.100
-- Checking for module 'libavresample'
--   Found libavresample, version 3.7.0
-- Checking for module 'gstreamer-base-1.0'
--   Found gstreamer-base-1.0, version 1.14.5
-- Checking for module 'gstreamer-app-1.0'
```
It gets stuck at checking 'gstreamer-app-1.0' module (more than 1 hour). 
Please help.
Thanks

mdegans, 2020-04-18:
Sorry for the late response. I had a deadline this week.

That's strange. That could be a lot of things. Have you run `top` or similar to monitor CPU / disk / memory usage during the build? That would be useful. On the Nano devkit you could try a new SD card.

mdegans, 2020-04-24:
haven't heard back, so i'm gonna close this, but feel free to re-open it if you're still having the issue.

closed 2020-04-24 by mdegans