mdegans/nano_build_opencv #32: Trying to get built opencv on jetson nano to work with virtual environment

aditjha, 2020-06-12:
Hello, thank you for your build script for the nano. I ran the script overnight and I think it finished building, however is there a way to check if it built successfully? And is there any way you know how to get this built opencv to work with a virtual environment on the nano, i ran the script from inside the virtual environment but it still doesn't find module cv2. Help would be super appreciated thank you!

currently in my /tmp/ folder i see a folder build-opencv, however I don't know if it is supposed to be like that.

mdegans, 2020-06-12:
Hi, @aditjha 

To **check for a sucessful install**:
```text
$ opencv_version --verbose
```

You should see something like:
```text
 $ opencv_version --verbose

General configuration for OpenCV 4.3.0 =====================================
  Version control:               4.3.0

  Extra modules:
    Location (extra):            /tmp/build_opencv/opencv_contrib/modules
    Version control (extra):     4.3.0

  Platform:
    Timestamp:                   2020-05-14T20:55:02Z
    Host:                        Linux 4.9.140-tegra aarch64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE
...
```

**To check if the python module is installed**:
```text
 $ python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
v>>> cv2.__version__
'4.3.0'
```

**To check if GPU support works in python**:
```text
>>> cv2.cuda.printCudaDeviceInfo(0)
*** CUDA Device Query (Runtime API) version (CUDART static linking) *** 

Device count: 1

Device 0: "Xavier"
  CUDA Driver Version / Runtime Version          10.20 / 10.20
  CUDA Capability Major/Minor version number:    7.2
  Total amount of global memory:                 15815 MBytes (16582963200 bytes)
  GPU Clock Speed:                               1.38 GHz
  Max Texture Dimension Size (x,y,z)             1D=(131072), 2D=(131072,65536), 3D=(16384,16384,16384)
  Max Layered Texture Size (dim) x layers        1D=(32768) x 2048, 2D=(32768,32768) x 2048
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 65536
  Warp size:                                     32
  Maximum number of threads per block:           1024
  Maximum sizes of each dimension of a block:    1024 x 1024 x 64
  Maximum sizes of each dimension of a grid:     2147483647 x 65535 x 65535
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and execution:                 Yes with 1 copy engine(s)
  Run time limit on kernels:                     No
  Integrated GPU sharing Host Memory:            Yes
  Support host page-locked memory mapping:       Yes
  Concurrent kernel execution:                   Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support enabled:                No
  Device is using TCC driver mode:               No
  Device supports Unified Addressing (UVA):      Yes
  Device PCI Bus ID / PCI location ID:           0 / 0
  Compute Mode:
      Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) 

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version  = 10.20, CUDA Runtime Version = 10.20, NumDevs = 1
```

**Please note** you must use `cv2.cuda.foo` instead of `cv2.foo` if you want your code to use the GPU. This means **most code for opencv you find on the internet must be rewritten** to use equivalents in the `cuda` module (if they exist) **if you want GPU acceleration**. Otherwise your code will use the CPU exclusively (OpenCV is an Intel supported product) and this will be very very slow on an Arm CPU like Nano.

A **virtual environment** will not include system python packages unless you specify it on creation with an option like `--system-site-packages` in the case of `virtualenv`.  If for some reason this doesn't work you should be able to copy or link opencv from it's installation path to your virtual env's python path. Like:

```
ln -s /usr/local/lib/python3.6/dist-packages/cv2/ /path/to/venv/lib/python3.6/site-packages/
```
(this is untested personally but should work)

**Easier** if you want repeatability might be the **docker images** [here](https://hub.docker.com/r/mdegans/tegra-opencv/tags). To get gpu support you must run with the `runtime --nvidia` flag when you `docker run`. There are tags for JetPack 4.3(jp-r32.3.1...) as well as 4.4 (jp-r32.4.2...).

Hope that answers your questions!

Sreerag-ibtl, 2020-10-16:
I tried the symlink but it didn't worked. So, I copied the directory cv2 from usr/local/lib/python3.6/site-packages to [path to venv]/libpython3.6/site-packages and that worked.

beniroquai, 2022-01-25:
Hey, thanks for the information! Did anybody try that with a miniconda/miniforge installation? 

After installing it using your script and linking it I get the following error when importing `cv2`:

```
(imswitch) uc2@uc2-desktop:~/miniforge3/envs/imswitch/lib/python3.7$ python
Python 3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 05:43:52) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
['', '/home/uc2/miniforge3/envs/imswitch/lib/python3.7/cv2/python-3.7', '/home/uc2/miniforge3/envs/imswitch/lib/python37.zip', '/home/uc2/miniforge3/envs/imswitch/lib/python3.7', '/home/uc2/miniforge3/envs/imswitch/lib/python3.7/lib-dynload', '/home/uc2/miniforge3/envs/imswitch/lib/python3.7/site-packages', '/home/uc2/Downloads/imswitch']
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/uc2/miniforge3/envs/imswitch/lib/python3.7/cv2/__init__.py", line 96, in <module>
    bootstrap()
  File "/home/uc2/miniforge3/envs/imswitch/lib/python3.7/cv2/__init__.py", line 86, in bootstrap
    import cv2
  File "/home/uc2/miniforge3/envs/imswitch/lib/python3.7/cv2/__init__.py", line 96, in <module>
    bootstrap()
  File "/home/uc2/miniforge3/envs/imswitch/lib/python3.7/cv2/__init__.py", line 23, in bootstrap
    raise ImportError('ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.')
ImportError: ERROR: recursion is detected during loading of "cv2" binary extensions. Check OpenCV installation.
>>> 
```

My cmake is slightly modified:
```
configure () {
    local CMAKEFLAGS="
        -D BUILD_EXAMPLES=OFF
        -D BUILD_opencv_python2=OFF
        -D BUILD_opencv_python3=ON
        -D CMAKE_BUILD_TYPE=RELEASE
        -D CMAKE_INSTALL_PREFIX=${PREFIX}
#        -D CUDA_ARCH_BIN=5.3,6.2,7.2
#        -D CUDA_ARCH_PTX=
#        -D CUDA_FAST_MATH=ON
#        -D CUDNN_VERSION='8.0'
        -D PYTHON3_EXECUTABLE=~/miniforge3/envs/imswitch/bin/python
        -D PYTHON3_LIBRARY=~/miniforge3/envs/imswitch/lib/python3.7m.dylib
        -D PYTHON3_INCLUDE_DIR=~/miniforge3/envs/imswitch/include/python3.7m
        -D PYTHON3_PACKAGES_PATH=~/miniforge3/envs/imswitch/lib/python3.7/site-packages
        -D EIGEN_INCLUDE_PATH=/usr/include/eigen3
        -D ENABLE_NEON=ON
#        -D OPENCV_DNN_CUDA=ON
        -D OPENCV_ENABLE_NONFREE=ON
        -D OPENCV_EXTRA_MODULES_PATH=/tmp/build_opencv/opencv_contrib/modules
        -D OPENCV_GENERATE_PKGCONFIG=ON
#        -D WITH_CUBLAS=ON
#        -D WITH_CUDA=ON
#        -D WITH_CUDNN=ONF
        -D WITH_GSTREAMER=ON
        -D WITH_LIBV4L=ON
        -D WITH_OPENGL=OFF"
```

Any ideas?