mdegans/nano_build_opencv #33: Running this script on Jetson Xavier NX makes it unbootable

iliesaya, 2020-06-17:
So I naively thought that this script will work on my Jetson Xavier NX , guess what, it's killed the boot. Why does it need to change my bootloader and my firmware to install openCV ? How can I fix that ? formating the SD card didn't change anything ...

here is the install log [https://pastebin.com/EywverjH](https://pastebin.com/EywverjH )

mdegans, 2020-06-17:
Before you panic, I don't think it's possible to permanently brick your NX (or any Tegra device).

To ensure repeatability, the script updates the system and installs some packages, **none of which include a bootloader or anything having to do with the boot flow** (read the script if you don't believe me). It looks like your apt configuration (eg. apt lists, foreign architectures) _could_ what may have rendered your system **temporarily** unbootable, not the script. It would likely have broken at the next update anyway.

I would recommend against setting your system up the same way again. If I had to guess it would have something to do with how TeamViewer was set up. I'd recommend not using that. It is **unsupported on aarch64** and setting up foreign architectures has the potential to cause issues like this.

My suggestion if you hadn't already formatted the card would be to back up your files using a SD card reader and a Linux desktop, but that is not possible after a format of the disk.

My suggestion **to fix your problem** would be to **follow [these instructions](https://developer.nvidia.com/embedded/learn/get-started-jetson-xavier-nx-devkit)** to reflash your SD card and reboot (formatting the card just erases your data).

Please note SDK manager is not required to install any software anymore and I would strongly recommend against it's use in favor of OTA updates. SDK manager installs local apt repositories that take up space and can conflict with OTA packages.

iliesaya, 2020-06-17:
Dear @mdegans  , thank you very much for trying to help, I do appreciate your support.
I written an new image (Jetson Xavier NX Developer Kit SD Card Image) on a brand new micro SD card , and I am still stuck on the same screen when I boot the jetson Xavier NX 
"bash: cannot set terminal process group (-1): Inappropriate ioctl for device" , so I believe my only solution is to re-flash with jetpack. The problem is that I don't have a computer under Ubuntu (I have a MacBook pro) , virtual machine are failing the flashing step ... don't know what to try :(

mdegans, 2020-06-18:
> Dear @mdegans , thank you very much for trying to help, I do appreciate your support.
> I written an new image (Jetson Xavier NX Developer Kit SD Card Image) on a brand new micro SD card , and I am [...]

My advice would be to ask [here](https://forums.developer.nvidia.com/c/agx-autonomous-machines/jetson-embedded-systems/jetson-xavier-nx/), and a Nvidia representative can give you an authoritative answer (or tell you if you have a hardware problem). I just got my NX and so my undestanding what's possible is more limited than with other boards, but I do not believe my script was a (direct) cause of your error. Still, I would like to know if I am wrong becuase if my script did trigger the issue, I want it to modify it so it doesn't happen to others in this particular case. Strange things do happen from time to time.

If you do post a thread, please link to it here so I can follow along.

I did find a thread with [your same error](https://forums.developer.nvidia.com/t/jetson-wont-load-up-after-configuring-spi-functionality/112139) on tx2. You might have a look and see if it applies to you as well.

As far as reflashing, yes, virtual machines will not work since the device is ejected midway through the process. Manually using flash.sh _may_ work, however, so you could also update in that thread and ask for the advice of linuxdev who is very familiar with this tool. That being said, it's my undestanding it's just using your board as a sd card writer, so I'm not sure it'll help with devices that have sd cards.

iliesaya, 2020-06-18:
here is the discussion on nvidia forum https://forums.developer.nvidia.com/t/failed-to-boot-cpu-bootloader-is-not-running-on-device/128323/11

mdegans, 2020-07-09:
@iliesaya 
I had this error myself at a recent update, FYI. I believe Nvidia is working on it per this post.
https://forums.developer.nvidia.com/t/jetpack-4-4-l4t-r32-4-3-production-release/140866

Fwiw, [It fixed itself after several reboots for me](https://forums.developer.nvidia.com/t/tx1-move-filesystem-to-sd-l4t-r32-rev-2-3/108947/15?u=mdegans). I have no idea how or why, but it boots now. I'm gonna close this as it appears confirmed as unrelated to the script at this point (it happend to me after an update, but not after the script). If you feel it is related, however, feel free to re-open this.

closed 2020-07-09 by mdegans

mdegans, 2020-07-10:
@iliesaya 

It's confirmed that it's an issue with the bootloader update pushed by Nvidia. You were one of the first people to have this issue, if not the first. Fix instructions [are here](https://forums.developer.nvidia.com/t/jetson-xavier-nx-not-booting/141070/27) for anybody who might google and find this thread.