mdegans/nano_build_opencv #34: I think your dockerhub image for r32.3.1 is built on linux/amd64

lackdaz, 2020-06-30:
https://hub.docker.com/layers/mdegans/tegra-opencv/jp-r32.3.1-cv-4.3.0/images/sha256-dd1bf2da56d18f497bf0882645d2120b0765fe53680615fdeebcf38dcfea5c4c?context=explore

Shouldn't you have compiled it on arm64? I'm assuming you are compiling it on buildx or qemu

That said, I'm appreciative of the script. The jetsonhacks one is getting dated and doesn't work for docker container builds.

mdegans, 2020-06-30:
I'm not sure why the arch meta is wrong on Docker Hub. It's built on a Tegra device I use for the purpose. I think the issue is with the base image of JetPack 4.3. I would guess the same cause you did (qemu-user-static etc...). The 4.4 meta is correct, so it appears Nvidia fixed it. If for some reason `mdegans/tegra-opencv:jp-r32.3.1-cv-4.3.0` _doesn't_ run on Tegra, feel free to re-open this.

closed 2020-06-30 by mdegans

lackdaz, 2020-06-30:
I'm trying to find a docker image/build that details a build that runs with opencv

I've tried yours and got the dreaded segmentation fault (core dumped) error *sigh*

Steps to reproduce:
`docker run -it mdegans/tegra-opencv:jp-r32.3.1-cv-4.3.0 /bin/bash`
`python3`
`import cv2`

I'm trying to run it on a nano with r32.3.1 and the reason why I'm doing it in docker containers is because I got a docker-compose microservices approach going.

I've actually been dancing around this issue for some time and its quite vexing. Do you have a personal email/contact I could use? I've read your code and seen a lot of your solutions on the devtalk forums and would like to engage you on a professional basis (or at least contribute something) for some deploy ops, but haven't found a point of contact. I can be contacted on my <username>@gmail.com

mdegans, 2020-06-30:
That can happen when you don't specify `--runtime nvidia` when you `docker run`.

Example run:
```
(sudo) docker run --user $(id -u):$(cut -d: -f3 < <(getent group video)) --runtime nvidia -it --rm mdegans/tegra-opencv:jp-r32.3.1-cv-4.3.0
```
Depending on the compose version, you may not be able to use `runtime`. I think 2.x supports it while 3.x does not. In 3.x you can specify gpus as resources (only with swarm), but this is not available for the tegra nvidia runtime yet.

mdegans, 2020-06-30:
@lackdaz Re: contact: michael.john.degans@gmail.com

lackdaz, 2020-06-30:
Ok, I hang my head in shame because I've made this error before. I probably should get the solution tattooed

I'll be contacting you

mdegans, 2020-06-30:
> Ok, I hang my head in shame because I've made this error before.

@lackdaz 
No worries. It happens to me all the time. You're not the first to have that error. I put the example run in the readme, but I think a lot of people skip it. I should probably add an entrypoint that checks for the gpu device and throws a descriptive error if it's not mounted.