mdegans/nano_build_opencv #35: Failure to build on Jetpack 4.4

JVTEAM, 2020-07-10:
[ 50%] Building CXX object modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o
In file included from /tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn.hpp:8:0,
                 from /tmp/build_opencv/opencv/modules/dnn/src/layers/../op_cuda.hpp:11,
                 from /tmp/build_opencv/opencv/modules/dnn/src/layers/blank_layer.cpp:43:
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp: In constructor ‘cv::dnn::cuda4dnn::csl::cudnn::ConvolutionAlgorithm<T>::ConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&)’:
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:266:21: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ was not declared in this scope
                     CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:266:21: note: suggested alternative: ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’
                     CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp: In constructor ‘cv::dnn::cuda4dnn::csl::cudnn::TransposeConvolutionAlgorithm<T>::TransposeConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&)’:
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp:42:21: error: ‘CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST’ was not declared in this scope
                     CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp:42:21: note: suggested alternative: ‘CUDNN_CONVOLUTION_BWD_DATA_ALGO_FFT’
                     CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
modules/dnn/CMakeFiles/opencv_dnn.dir/build.make:721: recipe for target 'modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o' failed
make[2]: *** [modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o] Error 1
CMakeFiles/Makefile2:3798: recipe for target 'modules/dnn/CMakeFiles/opencv_dnn.dir/all' failed
make[1]: *** [modules/dnn/CMakeFiles/opencv_dnn.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2

mdegans, 2020-07-10:
I can replicate with JetPack 4.4 GA

Problem [seems related to deprecated functions removed from cuDNN 8.0](https://forums.developer.nvidia.com/t/jetpack-4-4-broke-one-of-my-programs/141193/17?u=mdegans) (thanks @dusty-nv )
```
In file included from /tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn.hpp:8:0,
                 from /tmp/build_opencv/opencv/modules/dnn/src/layers/../op_cuda.hpp:11,
                 from /tmp/build_opencv/opencv/modules/dnn/src/layers/blank_layer.cpp:43:
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp: In constructor 'cv::dnn::cuda4dnn::csl::cudnn::ConvolutionAlgorithm<T>::ConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&)':
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:266:21: error: 'CUDNN_CONVOLUTION_FWD_PREFER_FASTEST' was not declared in this scope
                     CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro 'CUDA4DNN_CHECK_CUDNN'
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:266:21: note: suggested alternative: 'CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3'
                     CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro 'CUDA4DNN_CHECK_CUDNN'
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp: In constructor 'cv::dnn::cuda4dnn::csl::cudnn::TransposeConvolutionAlgorithm<T>::TransposeConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor<T>&)':
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp:42:21: error: 'CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST' was not declared in this scope
                     CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro 'CUDA4DNN_CHECK_CUDNN'
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/transpose_convolution.hpp:42:21: note: suggested alternative: 'CUDNN_CONVOLUTION_BWD_DATA_ALGO_FFT'
                     CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
                     ^
/tmp/build_opencv/opencv/modules/dnn/src/layers/../cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro 'CUDA4DNN_CHECK_CUDNN'
     ::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, __FILE__, __LINE__)
                                                     ^~~~
modules/dnn/CMakeFiles/opencv_dnn.dir/build.make:721: recipe for target 'modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o' failed
make[2]: *** [modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```
I will try and see if a newer version of OpenCV builds and update here. Worst case scenario, it'll have to be build without cuDNN.

mdegans, 2020-07-10:
@JVTEAM

I am starting a build from master. If it's fixed upstream, I'll push the build to docker hub, give you the tag, and update the script so it builds outside docker. If not, i'll disable cuDNN until it's fixed.

dusty-nv, 2020-07-10:
There are a few PRs / open issues I see on OpenCV github for cuDNN 8 support.  Here is one - https://github.com/opencv/opencv/pull/17685

mdegans, 2020-07-10:
thx @dusty-nv 

Looks like whatever was breaking the dnn module is fixed in master already. The configuration says cuDNN is enabled and the module built, so good news. Thx again for the insight on the error.
```
[ 70%] Linking CXX shared library ../../lib/libopencv_dnn.so
[ 70%] Built target opencv_dnn
```
@JVTEAM You can try `./build_opencv.sh master` and see if it works for you. I am building in docker with a slightly different script, but there's a good chance it will work. In any case, I will test tonight and update when complete.

mdegans, 2020-07-11:
@JVTEAM 

It looks like master does build and cuda support is available. I did not test the cuDNN module but it did build. If you use it, please report any issues you find on github. The docker image has just been pushed. You may:

```
sudo docker run -it --rm --runtime nvidia mdegans/tegra-opencv:jp-r32.4.3-cv-master
``` 
(it's also the "latest" tag)
and within the container, run
```
root@c0a37a2a0bd4:/usr/local/src/build_opencv# python3
Python 3.6.9 (default, Apr 18 2020, 01:56:04) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> cv2.cuda.printCudaDeviceInfo(0)
*** CUDA Device Query (Runtime API) version (CUDART static linking) *** 

Device count: 1

Device 0: "Xavier"
  CUDA Driver Version / Runtime Version          10.20 / 10.20
  CUDA Capability Major/Minor version number:    7.2
  Total amount of global memory:                 7764 MBytes (8140648448 bytes)
  GPU Clock Speed:                               1.11 GHz
  Max Texture Dimension Size (x,y,z)             1D=(131072), 2D=(131072,65536), 3D=(16384,16384,16384)
  Max Layered Texture Size (dim) x layers        1D=(32768) x 2048, 2D=(32768,32768) x 2048
  Total amount of constant memory:               65536 bytes
  Total amount of shared memory per block:       49152 bytes
  Total number of registers available per block: 65536
  Warp size:                                     32
  Maximum number of threads per block:           1024
  Maximum sizes of each dimension of a block:    1024 x 1024 x 64
  Maximum sizes of each dimension of a grid:     2147483647 x 65535 x 65535
  Maximum memory pitch:                          2147483647 bytes
  Texture alignment:                             512 bytes
  Concurrent copy and execution:                 Yes with 1 copy engine(s)
  Run time limit on kernels:                     No
  Integrated GPU sharing Host Memory:            Yes
  Support host page-locked memory mapping:       Yes
  Concurrent kernel execution:                   Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support enabled:                No
  Device is using TCC driver mode:               No
  Device supports Unified Addressing (UVA):      Yes
  Device PCI Bus ID / PCI location ID:           0 / 0
  Compute Mode:
      Default (multiple host threads can use ::cudaSetDevice() with device simultaneously) 

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version  = 10.20, CUDA Runtime Version = 10.20, NumDevs = 1
```
Note: to use the gpu you do not need root, but your user needs to be mapped to or in the "video" group.

If you wish to build it yourself `./build_opencv.sh master` should work fine. Please re-open if it doesn't.

closed 2020-07-11 by mdegans