mdegans/nano_build_opencv #36: Build failed on Jetson Nano R32.4.2

qubitqualia, 2020-07-11:
[image]

mdegans, 2020-07-11:
Appears same issue as #35 

Problem can be solved by running
```
build_opencv.sh master
```

I will leave this open since a lot of people will likely have this problem soon. If people think I should change the default branch to master, upvote this comment.

mdegans, 2020-07-22:
@qubitqualia 
I'm going to close this since OpenCV 4.4.0 has been tagged upstream and builds. I've set it as the default version so `build_opencv.sh` should be all that's required. If you need any more help on this issue, feel free to re-open or comment on #35

closed 2020-07-22 by mdegans