mdegans/nano_build_opencv #37: Jetson shutting itself down when running code.

VegardIversen, 2020-07-13:
Thank you for build script. 
It seemed to work fine, but now when I run more heavier python detection code with opencv it shuts it self down. Anyone else with this problem or know a fix? 
the code im running worked fine when i had opencv 3.1.
The nano is in 5W mode, and im using a Raspberry Pi camera v2.

mdegans, 2020-07-13:
@VegardIversen 

This seems power supply related. I have not found a good Micro USB power supply for 5w mode other than the battery suggested for jetbot, and I tested a lot of MicroUSB power supplies. Shutdown/reboot seems to be the usual case, especially with connected peripherals. I would recommend retrying in 10w mode with a power adapter like this:

https://www.amazon.com/dp/B01N4HYWAM

That's the one I use. Other suggestions can be found on the Nvidia forum.

Why it works with 3.1 I can't say for certain, especially without looking at your code, but it could be 3.1 is not using the GPU at all, and with 4.x.x it is. In my own experience that's what cause the reboots: using the GPU and CPU at the same time. That was with some games however, and not OpenCV. I will leave this open for a few days in case other people are having this problem, but I don't think it's related to the script.

VegardIversen, 2020-07-13:
Hi @mdegans  thanks for answer.

Im using the battery suggested for jetbot. Im working with the sparkfun jetbot kit, so running a robot with  a power adapter might be difficult. I used face_recognition lib, and it max out the gpu, and used around 30-50 % (if I remember correctly) on the cpu with no problem. When I tried that it was with opencv 3.x (used for displaying the video and haarcascsade detection) and the Leopard Imaging 145 FOV.

But it cant be anything wrong with the camera settings? fps/size etc. 

Here is one of the codes that crashes. Its a bit messy.
```
from jetbot import Camera
from jetbot import bgr8_to_jpeg
from jetbot import Robot
import cv2
import os

import time
import subprocess

camera = Camera.instance(width=224, height=224)
#camera = Camera.instance()

#robot = Robot()
#robot.stop()

face_seen = 1

face_haar_model = './data/haarcascade_frontalface_default.xml'
smile_haar_model = './data/haarcascade_smile.xml'
eye_haar_model = './data/haarcascade_eye.xml'
face = cv2.CascadeClassifier(face_haar_model)
smile = cv2.CascadeClassifier(smile_haar_model)
eye = cv2.CascadeClassifier(eye_haar_model)
#face = cv2.CascadeClassifier(face_cascade)
#smile = cv2.CascadeClassifier(smile_cascade)

def drive(faces):
    if robot.forward():
        eyes = eye.detectMultiScale(roi_gray, 1.8, 5)
        if ((eyes != ()) and (time.perf_counter()-time_start > 50)):
            robot.stop()
    else:
        robot.forward()
        time_start = time.perf_counter()
        

print('yesy')
print(type(face))
def detect(gray, frame):
    global face_seen
    print('2.1')
    #faces = face_cascade.detectMultiScale(gray, 1.3, 5)
    faces = face.detectMultiScale(gray, 1.3, 2)
    print('2.2')
    for (x, y, w, h) in faces: 

        cv2.rectangle(frame, (x, y), ((x + w), (y + h)), (255, 0, 0), 2) 
        roi_gray = gray[y:y + h, x:x + w] 
        roi_color = frame[y:y + h, x:x + w] 
        print('2.3')
        #smiles = smile_cascade.detectMultiScale(roi_gray, 1.8, 20)
        eyes = eye.detectMultiScale(roi_gray, 1.8, 5)
        smiles = smile.detectMultiScale(roi_gray, 1.8, 5)
        print('2.4')
        if faces != ():
            if (face_seen != 1):
                
            
                print('its a face!')
            else:
                #subprocess.run(['play','./data/hello.mp3'])
                face_seen = 0
            #drive(faces)
        if smiles != ():
            
            print('Nice smile')
        if eyes != ():
            print('hi eyes')
        else:
            print('\n')
            
  
        for (sx, sy, sw, sh) in smiles: 
            print('2.5')
            cv2.rectangle(roi_color, (sx, sy), ((sx + sw), (sy + sh)), (0, 0, 255), 2)
            print('2.6')
            
        
        for (sx, sy, sw, sh) in eyes:
            cv2.rectangle(roi_color, (sx, sy), ((sx + sw), (sy + sh)), (0, 0, 255), 2)
            
    return frame

while True:
    frame = camera.value
    print('1')
    print(frame.shape)

    gray = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
    print('2')
    canvas = detect(gray, frame)
    print('3')
    cv2.imshow('Video', canvas)

    
    if cv2.waitKey(1) & 0xff == ord('q'):
        #robot.stop()
        break

    print('8')

camera.stop()
cv2.destroyAllWindows()
```

Im gonna test more on another device tomorrow so if I find something I will post it here.

mdegans, 2020-07-13:
@VegardIversen 

I don't see anything in your code that's using the GPU except maybe the cv2.imshow. For example, the cascade classifier that uses cuda is [cv2.cuda.CacadeClassifier](https://docs.opencv.org/master/d9/d80/classcv_1_1cuda_1_1CascadeClassifier.html). Likewise much of the other stuff probably has equivalents in the cuda namespace.

If you say the GPU is maxed out, I am not sure what's causing that. It may be the camera or the imshow. In any case, it still seems like a power issue, especially if the GPU _is_ maxed out. Do you get the same GPU usage when you use OpenCV 3.1? Let me know if it's stable when you test with another power supply / device tomorrow. If it works that way, then the problem is confimed.

Lastly, when it reboots, is it sudden, or do you get some sort of error message beforehand?

VegardIversen, 2020-07-13:
@mdegans Oh im sorry to confuse thats a diffrent code that uses the GPU. The code i sent doesnt use GPU. 
So the GPU usage has been as expected. But still shutsdown.

It doesnt really reboot, it just suddenly shutsdown and the green light disappears, with no warning. But the computer mouse still lights up. When i turn it back on i get: ```tegradc tegradc. dpdk enable lookup fail:-19 ```. before it starts up again (can aslo just turn of again here). 

Here is the face_recgonition lib code that makes the GPU max, and this worked fine before i installed opencv4.3. 
```
locations = face_recognition.face_locations(image, model=MODEL)

```

VegardIversen, 2020-07-14:
Hi
I have tested more now. Im testing on two different jetson nano's one with jetpack 4.2 and one with jetpack 4.3. And Im using Lithium Ion Battery Pack - 10Ah (3A/1A USB Ports) that comes with the sparkfun jetbot ai kit. Tried with both the raspberry pi v2 camera and  Leopard Imaging 145 FOV. 
Tried not to display the video. This made it take a bit longer before it turned it self off but not much. 
The Leopard camera also make it run a bit longer than the raspberry camera. 
Also, almost for every sudden shutdown the jetson gets harder to turn on and not just shutdown again. 

From this I guess it is the powersupply that are the issue, but that is a bit weird because it worked fine for almost a month with both powering the motors and the jetson nano. I will try and get my hands on a better and bigger power supply.

let me know if anyone has any other tips. Its not good having a robot that are connected to the wall

mdegans, 2020-07-14:
@VegardIversen 
It looks like that library (`face_recognition`) uses dlib, which might use one of serveral OpenCV versions (or none at all) depending on how it's built. (the [build requirements](http://dlib.net/compile.html) do not list it and i can't see it in the dlib/CmakeLists.txt). It does look like it can use OpenCV Mat to get images in and out, but the stuff that's making your GPU work hard is pure dlib.

If you're looking for more tips you might ask on the Nvidia developer forum. There's more people there and more eyes looking at a probem is _usually_ a good thing, especially when some of them are Nvidia core developers themselves. In any case, if you can plug your Nano into a barrel jack power supply (and jump j48), you'll be able to conclusively determine if the issue is power related or not. I'll leave this issue open until then, since there is still a chance it might be OpenCV related. More likely at this point, however, i'd say power supply, dlib, cuda, the drivers, or possibly even the kernel.

VegardIversen, 2020-07-18:
hi @mdegans 
Done alot of debugging. And to conclude it was a combined problemed between power issues and higher version of opencv. Using the the raspberry power adapter stopped the jetson shutting down. Also tried to use an older version of jetpack (4.2.1) with opencv 3.2.1. This made the jetson able to run with a battery pack, as long I didnt use to much GPU (it seems), so using face_recognition(dlib) made it crash. Still dont know why this worked earlier and not now, but guessing the quality of the battery has gone down.

mdegans, 2020-07-22:
> hi @mdegans
> Done alot of debugging. And to conclude it was a combined problemed between power issues and higher version of opencv. Using the the raspberry power adapter stopped the jetson shutting [...]

That's certainly possible, @VegardIversen . I'm glad you were able to confirm the problem. If you can test with a barrel jack power adapter, my guess is it'd be completely stable. I have not been able to find a reliable micro-usb power supply for Nano in 5w mode, personally. You might ask on the Nvidia forum whether anybody has any working recommendations. I am certain there are other people who run it in 5w mode.

I'm closing this issue as it seeems you've confirmed the issue as power related. If you continue having issues, feel free to re-open.

closed 2020-07-22 by mdegans

VegardIversen, 2020-07-23:
> That's certainly possible, @VegardIversen . I'm glad you were able to confirm the problem. If you can test with a barrel jack power adapter, my guess is it'd be completely stable. I have not been [...]

Hi, 
Raspberry Pi power supply seems to be reliable in 5W mode. Thank you for your input and help.