mdegans/nano_build_opencv #39: build failed, "repository no longer signed"

GOBish, 2020-07-14:
I have used your excellent script in the past successfully, but it seems something changed recently that is giving me the following error:

```
Reading package lists... Done                      
E: Failed to fetch http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo/dists/bionic/InRelease  403  Forbidden [IP: 52.218.53.57 80]
E: The repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

```

mdegans, 2020-07-14:
That would seem to be a problem with the with intel RealSense repository configuration. You should be able to fix that by registering the public key as shown in the setup docs here:

https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md#installing-the-packages

Feel free to re-open if that doesn't fix it, or you can google the error and there are a few issues about it on the realsense repo.

closed 2020-07-14 by mdegans

GOBish, 2020-07-16:
Thanks, got it to work