mdegans/nano_build_opencv #40: Couldnot build OpenCV DNN

H-Kz, 2020-07-18:
The same issue as [here](https://forums.developer.nvidia.com/t/i-cant-use-cuda-dnn-on-jetson-nano-python/142471/5).
Thank you very much for your help.

Now I'm trying to use cuDNN on my Jetson nano, but build is not successable.
I've eddited your build_opencv.sh just these 2 lines,

>111    -D CUDA_ARCH_BIN=5.3
>112    -D CUDA_ARCH_PTX=5.3

and I've ran this script.

> ./build_opencv.sh

Then I got these Output.

```
 [ 46%] Building CXX object modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o
In file included from /tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/csl/cudnn.hpp:8:0,
from /tmp/build_opencv/opencv/modules/dnn/src/layers/…/op_cuda.hpp:11,
from /tmp/build_opencv/opencv/modules/dnn/src/layers/blank_layer.cpp:43:
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/convolution.hpp: In constructor ‘cv::dnn::cuda4dnn::csl::cudnn::ConvolutionAlgorithm::ConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor&)’:
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/convolution.hpp:266:21: error: ‘CUDNN_CONVOLUTION_FWD_PREFER_FASTEST’ was not declared in this scope
CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
^
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, FILE, LINE)
^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/convolution.hpp:266:21: note: suggested alternative: ‘CUDNN_CONVOLUTION_BWD_FILTER_ALGO_3’
CUDNN_CONVOLUTION_FWD_PREFER_FASTEST,
^
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, FILE, LINE)
^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/transpose_convolution.hpp: In constructor ‘cv::dnn::cuda4dnn::csl::cudnn::TransposeConvolutionAlgorithm::TransposeConvolutionAlgorithm(const cv::dnn::cuda4dnn::csl::cudnn::Handle&, const cv::dnn::cuda4dnn::csl::cudnn::ConvolutionDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::FilterDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor&, const cv::dnn::cuda4dnn::csl::cudnn::TensorDescriptor&)’:
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/transpose_convolution.hpp:42:21: error: ‘CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST’ was not declared in this scope
CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
^
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, FILE, LINE)
^~~~
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/primitives/…/csl/cudnn/transpose_convolution.hpp:42:21: note: suggested alternative: ‘CUDNN_CONVOLUTION_BWD_DATA_ALGO_FFT’
CUDNN_CONVOLUTION_BWD_DATA_PREFER_FASTEST,
^
/tmp/build_opencv/opencv/modules/dnn/src/layers/…/cuda4dnn/csl/cudnn/cudnn.hpp:23:53: note: in definition of macro ‘CUDA4DNN_CHECK_CUDNN’
::cv::dnn::cuda4dnn::csl::cudnn::detail::check((call), CV_Func, FILE, LINE)
^~~~
modules/dnn/CMakeFiles/opencv_dnn.dir/build.make:7134: recipe for target ‘modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o’ failed
make[2]: *** [modules/dnn/CMakeFiles/opencv_dnn.dir/src/layers/blank_layer.cpp.o] Error 1
CMakeFiles/Makefile2:3737: recipe for target ‘modules/dnn/CMakeFiles/opencv_dnn.dir/all’ failed
make[1]: *** [modules/dnn/CMakeFiles/opencv_dnn.dir/all] Error 2
Makefile:162: recipe for target ‘all’ failed
make: *** [all] Error 2
Do you wish to remove temporary build files in /tmp/build_opencv ?
(Doing so may make running tests on the build later impossible)
Y/N n
Please answer yes or no.
Do you wish to remove temporary build files in /tmp/build_opencv ?
(Doing so may make running tests on the build later impossible)
Y/N n
```

This is output of opencv_version --verbose

```
$ opencv_version 4.4.0-pre opencv_version --verbose

General configuration for OpenCV 4.4.0-pre =====================================
Version control: 8a480ac

Extra modules:
[... 89 lines cut ...]
Include path: /tmp/build_opencv/opencv/3rdparty/include/opencl/1.2
Link libraries: Dynamic load

Python 2:
Interpreter: /usr/bin/python2.7 (ver 2.7.17)
Libraries: /usr/lib/aarch64-linux-gnu/libpython2.7.so (ver 2.7.17)
numpy: /usr/lib/python2.7/dist-packages/numpy/core/include (ver 1.13.3)
install path: lib/python2.7/dist-packages/cv2/python-2.7

Python 3:
Interpreter: /usr/bin/python3 (ver 3.6.9)
Libraries: /usr/lib/aarch64-linux-gnu/libpython3.6m.so (ver 3.6.9)
numpy: /usr/lib/python3/dist-packages/numpy/core/include (ver 1.13.3)
install path: lib/python3.6/dist-packages/cv2/python-3.6

Python (for build): /usr/bin/python2.7

Java:
ant: NO
JNI: NO
Java wrappers: NO
Java tests: NO

Install to: /usr/local
```

H-Kz, 2020-07-18:
Because it's midnight in Japan, I'll be afk for a few hours.
I7m sorry

H-Kz, 2020-07-18:
### Problem Solved!!

I ran your shell-scriot with specifyed version **4.4.0** , then building opencv with cuDNN succeed and it seems to be having no error!!

note: what I've changed is

1. edit these 2 line in build_opencv.sh

> 111     -D CUDA_ARCH_BIN=5.3
> 112     -D CUDA_ARCH_PTX=5.3

2.  Specify the OpenCV version 4.4.0

> ./build_opencv.sh 4.3.0

thank you for your grate help!!

mdegans, 2020-07-22:
Thanks for testing that @H-Kz 

I've made 4.4.0 the default version and made a note in the readme so `./build_opencv.sh` should be all that's required in the future.

closed 2020-07-22 by mdegans