mdegans/nano_build_opencv #43: Test failed with Jetpack 4.4 [L4T 32.4.3] and OpenCV 4.4.0

nalzok, 2020-09-13:
I'm building OpenCV 4.4.0 headlessly on a Jetson Nano with Jetpack 4.4 [L4T 32.4.3]. The build is successful but many tests failed. See attached logs.

[build.log](https://github.com/mdegans/nano_build_opencv/files/5213481/build.log)
[test.log](https://github.com/mdegans/nano_build_opencv/files/5213480/test.log)
[install.log](https://github.com/mdegans/nano_build_opencv/files/5213478/install.log)
[test-reports.zip](https://github.com/mdegans/nano_build_opencv/files/5213479/test-reports.zip)

Apparently the cause is multiple missing input files like `./gpu/opticalflow/frame0.png`. Any tips?

mdegans, 2020-09-13:
Thanks for reporting this @nalzok . The cause is that the data needed by the tests isn't downloaded when it should be if the test option is used. Many tests will still fail, but it's more like 80% passing, rather than 20% without the test data. I will implement a fix as soon as I get a chance.

More detail here:
https://docs.opencv.org/3.4/d6/d15/tutorial_building_tegra_cuda.html

see "OpenCV Extra"

If you clone that and set the approprate environment variable, it should find the test data.

nalzok, 2020-09-13:
Thanks for the timely response!

I tried to fix the issue myself by modifying your script to [this](https://github.com/mdegans/nano_build_opencv/files/5214432/build_opencv.sh.txt). I commented out the "prepare for the build" part and manually clone the `opencv`, `opencv_contrib` and `opencv_extra` repositories under the directory `/home/qingyao/opencv` instead, so that when something goes wrong I don't have to re-clone them again. The build script is also placed in `/home/qingyao/opencv`.

I have changed `CMAKEFLAGS` to enable more features, which may or may not have something to do with the error I got later. (Yeah I know that will cause more trouble for you to reproduce the issue. Sorry about that ):

The additional dependencies I installed (mostly for `-D BUILD_DOCS=ON`) are

```bash
sudo apt install doxygen graphviz openblas python-pip
pip install beautifulsoup4
```

I mounted a 4GB swap file and start the building process with `./build_opencv.sh foo test` (the first argument is ignore now). It took ~12 hours to finish on my Jetson Nano. This time, tests took longer to run, but many eventually still failed.

+ [build.log](https://github.com/mdegans/nano_build_opencv/files/5214449/build.log)
+ [test.log](https://github.com/mdegans/nano_build_opencv/files/5214448/test.log)
+ [install.log](https://github.com/mdegans/nano_build_opencv/files/5214450/install.log)

You may download the full test report [here](https://carrier.oss-cn-shanghai.aliyuncs.com/test-reports.zip). It's a 134.1 MB zip file including some image files.

There are a lot of different errors, but the "cuDNN did not return a suitable algorithm for convolution" error gets the most exposure. Here is an example of what it looks like

```
test-reports/accuracy/opencv_test_dnn.xml
8174:      <failure message="unknown file&#x0A;C++ exception with description &quot;OpenCV(4.4.0) /home/qingyao/opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:303: error: (-217:Gpu API call) cuDNN did not return a suitable algorithm for convolution. in function &apos;ConvolutionAlgorithm&apos;&#x0A;&quot; thrown in the test body." type=""><![CDATA[unknown file
8440:      <failure message="unknown file&#x0A;C++ exception with description &quot;OpenCV(4.4.0) /home/qingyao/opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:303: error: (-217:Gpu API call) cuDNN did not return a suitable algorithm for convolution. in function &apos;ConvolutionAlgorithm&apos;&#x0A;&quot; thrown in the test body." type=""><![CDATA[unknown file
8677:      <failure message="/home/qingyao/opencv/opencv/modules/dnn/test/test_common.impl.hpp:70&#x0A;Expected: (normL1) &lt;= (l1), actual: nan vs 1e-05" type=""><![CDATA[/home/qingyao/opencv/opencv/modules/dnn/test/test_common.impl.hpp:70
8684:      <failure message="/home/qingyao/opencv/opencv/modules/dnn/test/test_common.impl.hpp:70&#x0A;Expected: (normL1) &lt;= (l1), actual: nan vs 0.004" type=""><![CDATA[/home/qingyao/opencv/opencv/modules/dnn/test/test_common.impl.hpp:70
9230:      <failure message="unknown file&#x0A;C++ exception with description &quot;OpenCV(4.4.0) /home/qingyao/opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:303: error: (-217:Gpu API call) cuDNN did not return a suitable algorithm for convolution. in function &apos;ConvolutionAlgorithm&apos;&#x0A;&quot; thrown in the test body." type=""><![CDATA[unknown file
9335:      <failure message="unknown file&#x0A;C++ exception with description &quot;OpenCV(4.4.0) /home/qingyao/opencv/opencv/modules/dnn/src/layers/../cuda4dnn/primitives/../csl/cudnn/convolution.hpp:303: error: (-217:Gpu API call) cuDNN did not return a suitable algorithm for convolution. in function &apos;ConvolutionAlgorithm&apos;&#x0A;&quot; thrown in the test body." type=""><![CDATA[unknown file
```

------

However, the OpenCV installation *appears* good

```
qingyao@jetson:~/opencv/opencv/build$ opencv_version --verbose

General configuration for OpenCV 4.4.0 =====================================
  Version control:               4.4.0

  Extra modules:
    Location (extra):            /home/qingyao/opencv/opencv_contrib/modules
    Version control (extra):     4.4.0

  Platform:
    Timestamp:                   2020-09-13T04:59:57Z
    Host:                        Linux 4.9.140-tegra aarch64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON
      disabled:                  VFPV3

  C/C++:
    Built as dynamic libs?:      YES
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 7.5.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -O3 -DNDEBUG  -DNDEBUG
[... cut ...]
  Concurrent kernel execution:                   Yes
  Alignment requirement for Surfaces:            Yes
  Device has ECC support enabled:                No
  Device is using TCC driver mode:               No
  Device supports Unified Addressing (UVA):      Yes
  Device PCI Bus ID / PCI location ID:           0 / 0
  Compute Mode:
      Default (multiple host threads can use ::cudaSetDevice() with device simultaneously)

deviceQuery, CUDA Driver = CUDART, CUDA Driver Version  = 10.20, CUDA Runtime Version = 10.20, NumDevs = 1

>>> 
```

mdegans, 2020-12-10:
Sorry for the late response @nalzok.  Your issue may be that `OPENCV_TEST_DATA_PATH` (cmake) is not set. If you check the above link, there should be a line like:

```
    -DINSTALL_TESTS=OFF \
    -DOPENCV_TEST_DATA_PATH=../opencv_extra/testdata \
```

I will be pushing some updates today to fix this issue, hopefully. Thanks again for reporting it.

commit 61f79f89 (2020-12-11): fix #43 WIP

This does not fix the issue, but i'm not sure why since OPENCV_TEST_DATA_PATH
is defined. A second pair of eyes might help