mdegans/nano_build_opencv #44: OpenCV was not built with xfeatures2d module

AlexPolg, 2020-09-17:
Hello, i built opencv using your instruction and when using rtabmap the following errors appear.  Features2d.cpp:529::create() BRIEF and FREAK features cannot be used because OpenCV was not built with xfeatures2d module. GFTT/ORB is used instead.   Features2d.cpp:1084::parseParameters() GPU version of ORB not available (OpenCV cudafeatures2d module)! Using CPU version instead... please help me how to install or fix this module

mdegans, 2020-09-21:
That should be built since `OPENCV_ENABLE_NONFREE=ON` is the default in the script. I will check it out. Thanks for reporting.

AlexPolg, 2020-09-23:
Thank you, I will wait for your answer

AlexPolg, 2020-09-23:
Or maybe the problem is in the opencv version? i installed 4.4.0

SpikeYZY, 2024-06-05:
get any solution？