mdegans/nano_build_opencv #46: error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.

cxhmikegithub, 2020-09-25:
### I got this error:
Getting version '4.0.0' of OpenCV
Cloning into 'opencv'...
remote: Enumerating objects: 6852, done.
remote: Counting objects: 100% (6852/6852), done.
remote: Compressing objects: 100% (5928/5928), done.
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

mdegans, 2020-10-08:
Seems like a connection issue @cxhmikegithub

Please try again. In these times, internet connections are stressed and this thing can happen more frequently. Please feel free to re-open if the issue persists.

closed 2020-10-08 by mdegans

HXB-1997, 2023-11-01:
> Seems like a connection issue @cxhmikegithub
> 
> Please try again. In these times, internet connections are stressed and this thing can happen more frequently. Please feel free to re-open if the issue persists.
I also meet this problem:
Getting version '4.8.0' of OpenCV
正克隆到 'opencv'...
fatal: unable to access 'https://github.com/opencv/opencv.git/': Failed to connect to github.com port 443: 连接超时
@mdegans