mdegans/nano_build_opencv #47: NVCUVID support

estebanroblesluna, 2020-10-30:
Hi Michael,
First of all, thanks for sharing your knowledge and this script.
I have been using OpenCV with CUDA in a Bare metal server and now I'm trying to use it in a Jetson Nano.
I have modified your script to support NVCUVID by adding the flag:
-D WITH_NVCUVID=ON

However, when I execute the script, it doesn't show that is building using it:

NVIDIA CUDA: YES(ver 10.2 CUFFT CUBLAS FAST_MATH)

Do you have any insights on why this may be happening?
Regards

mdegans, 2020-10-30:
Seems it is unsupported:

https://github.com/opencv/opencv/issues/8445

Not sure if it ever was on Tegra. If you want hardware decode/encode, your best bet is probably using a GStreamer pipeline in VideoCapture/VideoWriter.

Hope that helps, @estebanroblesluna !

closed 2020-10-30 by mdegans

estebanroblesluna, 2020-10-30:
Hi @mdegans, I tried it but the CPU level goes to 20%, which is something I'm not expecting. Is that normal? It is a RTSP stream of 1080p
Regards