mdegans/nano_build_opencv #49: jetson nano Nvidia failed in build

gadididi, 2020-12-23:
i have:
**jetson nano Nvidia (arm64 architecture) with ububntu 18**
i tried to run this script ( i need CUDA in opencv for run codes on GPU)
i made a little change in the script -rename the name "build_opencv" to "buildopencv" ( it is not the issue )
i run the script ./build_opencv.sh 4.3.0 
and i got failed in the building ...
i attached the cmake log output and cmake error log:
[CMakeError.log](https://github.com/mdegans/nano_build_opencv/files/5735149/CMakeError.log)
[CMakeOutput.log](https://github.com/mdegans/nano_build_opencv/files/5735150/CMakeOutput.log)

if someone can help me ,please .. :)

gadididi, 2020-12-27:
solved. 
my jeckpack nano was invalid

closed 2020-12-27 by gadididi

wgthompson, 2023-06-09:
> solved. my jeckpack nano was invalid

How exactly was it solved? Thanks