mdegans/nano_build_opencv #52: Errors were encountered while processing:  nvidia-l4t-kernel-dtbs

prateek-gk, 2021-03-10:
I am trying to build on NX but build fails:

```
It appears an existing build exists in /tmp/build_opencv
Do you wish to remove temporary build files in /tmp/build_opencv ? 
(Doing so may make running tests on the build later impossible)
Y/N Y
Installing build dependencies.
Get:1 file:/var/cuda-repo-10-2-local-10.2.89  InRelease
Ign:1 file:/var/cuda-repo-10-2-local-10.2.89  InRelease
Get:2 file:/var/visionworks-repo  InRelease
Ign:2 file:/var/visionworks-repo  InRelease
Get:3 file:/var/visionworks-sfm-repo  InRelease
Ign:3 file:/var/visionworks-sfm-repo  InRelease
Get:4 file:/var/visionworks-tracking-repo  InRelease
Ign:4 file:/var/visionworks-tracking-repo  InRelease         
Get:5 file:/var/cuda-repo-10-2-local-10.2.89  Release [574 B]
Get:6 file:/var/visionworks-repo  Release [2,001 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                        
Get:5 file:/var/cuda-repo-10-2-local-10.2.89  Release [574 B]                  
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                   
Get:6 file:/var/visionworks-repo  Release [2,001 B]                            
Get:7 file:/var/visionworks-sfm-repo  Release [2,005 B]                        
Get:8 file:/var/visionworks-tracking-repo  Release [2,010 B]                   
Hit:9 https://repo.download.nvidia.com/jetson/common r32.5 InRelease           
Hit:10 https://repo.download.nvidia.com/jetson/t194 r32.5 InRelease            
Hit:11 http://packages.microsoft.com/repos/code stable InRelease               
Hit:12 http://packages.ros.org/ros/ubuntu bionic InRelease                     
Hit:14 http://ports.ubuntu.com/ubuntu-ports bionic InRelease                   
Get:15 http://ports.ubuntu.com/ubuntu-ports bionic-updates InRelease [88.7 kB]
Get:18 http://ports.ubuntu.com/ubuntu-ports bionic-backports InRelease [74.6 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports bionic-security InRelease [88.7 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports bionic-updates/main arm64 DEP-11 Metadata [290 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports bionic-updates/universe arm64 DEP-11 Metadata [284 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports bionic-backports/universe arm64 DEP-11 Metadata [9,288 B]
Get:24 http://ports.ubuntu.com/ubuntu-ports bionic-security/main arm64 DEP-11 Metadata [42.8 kB]
Get:25 http://ports.ubuntu.com/ubuntu-ports bionic-security/universe arm64 DEP-11 Metadata [55.4 kB]
Fetched 933 kB in 5s (202 kB/s)                                              
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up nvidia-l4t-kernel-dtbs (4.9.201-tegra-32.5.1-20210219084708) ...
Decompile /boot/dtb/kernel_tegra194-p3668-all-p3509-0000.dtb
tegra194-p3668-all-p3509-0000.dtb failed, quit.
dpkg: error processing package nvidia-l4t-kernel-dtbs (--configure):
 installed nvidia-l4t-kernel-dtbs package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 nvidia-l4t-kernel-dtbs
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

mdegans, 2021-04-12:
Looks related to update and a driver package on your system, not the script. Try commenting out the `apt-get update` and `apt-get upgrade` lines of the script and it should work, but be aware, there's likely a separate problem with your system. Do you have some third party driver packages installed?

mdegans, 2021-05-27:
gonna close this (unless there's more to add?)

closed 2021-05-27 by mdegans