mdegans/nano_build_opencv #54: Jetson nano 16 gb sd card.

remiruzn, 2021-04-12:
How can i install openCV with cuda support on Jetson nano with 16 gb sd card inside? If i try use script not enough memory error appear.

mdegans, 2021-04-12:
Try building without the gui on. To switch to the command line:

Temporary:
```
sudo systemctl isolate multi-user.target
```
Persistent across reboots:
```
sudo systemctl set-default multi-user.target
```
Once the build is done, just substitute `graphical.target` instead of `multi-user.target` to go back.

remiruzn, 2021-04-13:
`
[ 43%] Built target opencv_alphamat
[ 43%] Linking CXX shared library ../../lib/libopencv_cudafilters.so
[ 45%] Built target opencv_cudafilters
[ 46%] Building NVCC (Device) object modules/cudaimgproc/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_bilateral_filter.cu.o
<built-in>: fatal error: when writing output to /tmp/tmpxft_000078b9_00000000-5_bilateral_filter.cpp4.ii: No space left on device
compilation terminated.
CMake Error at cuda_compile_1_generated_bilateral_filter.cu.o.RELEASE.cmake:221 (message):
  Error generating
  /tmp/build_opencv/opencv/build/modules/cudaimgproc/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_bilateral_filter.cu.o

modules/cudaimgproc/CMakeFiles/opencv_cudaimgproc.dir/build.make:337: recipe for target 'modules/cudaimgproc/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_bilateral_filter.cu.o' failed
make[2]: *** [modules/cudaimgproc/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_bilateral_filter.cu.o] Error 1
CMakeFiles/Makefile2:3695: recipe for target 'modules/cudaimgproc/CMakeFiles/opencv_cudaimgproc.dir/all' failed
make[1]: *** [modules/cudaimgproc/CMakeFiles/opencv_cudaimgproc.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Do you wish to remove temporary build files in /tmp/build_opencv ? 
`
**Building without gui.**
Can i build from jetson with 16 gb sd card inside from flash card pluged in usb port?

mdegans, 2021-05-27:
@remiruzn 

You could add a swap file on an external disk, even usb, sure, but it will probably build slower than just turning the GUI off temporarily. Swap is slow on fast SSDs. MicroSD cards are generally even slower than spinning disks, so it'll probably hurt performance if you set that up.

closed 2021-05-27 by mdegans