mdegans/nano_build_opencv #55: OpenCV bindings for Python 3 not installed

laclouis5, 2021-05-20:
I installed OpenCV 4.5.2 via this script on a fresh Jetson Nano and checked the installation with the procedure described in this [issue](https://github.com/mdegans/nano_build_opencv/issues/32). This was working properly, Cuda was enabled and OpenCV version was 4.5.2 both using command `opencv_version` and in a Python 3 interpreter.

However, I restarted the Jetson Nano and now in a Python 3 interpreter I got version 4.1.1. If I open an interpreter in sudo mode it still shows version 4.5.2.

Do you know the cause of this issue? I thinks that solving it shouldn't be hard, just have to link the site-package version of OpenCV used for system python to the other site-package I guess.

laclouis5, 2021-05-20:
Ok, this did the trick:

```shell
sudo cp /usr/local/lib/python3.6/dist-packages/cv2/python-3.6/*.so /usr/lib/python3.6/dist-packages/cv2/python-3.6/
```

mdegans, 2021-05-20:
Thanks for spotting this. I'd consider it still an open issue. Looks like it might be a good idea to purge any existing opencv before installing since it'll likely use the old version.

mdegans, 2021-05-20:
Jsut a fyi, to anybody experiencing this, the above fix will work, but removing these apt package will also do the trick and possibly avoid other issues:
```
libopencv          libopencv-dev      libopencv-python   libopencv-samples
```
so `sudo apt purge` or `sudo apt remove` those and you should be fine.

Randy37773, 2022-06-27:
unfortunately this does not the trick for me.
build_opencv.sh runs without errors (4.4 and 4.5.5), but cuda-support is only available in python2.7 not in python3.6. 

> ```shell
> sudo cp /usr/local/lib/python3.6/dist-packages/cv2/python-3.6/*.so /usr/lib/python3.6/dist-packages/cv2/python-3.6/
> ```

tsubasa9647, 2022-10-20:
@Randy37773 
I was able to resolve the issue by running the script below. 
sudo cp /usr/local/lib/python3.6/site-packages/cv2/python-3.8/*.so /usr/lib/python3.8/dist-packages/cv2/python-3.8/

IraeMare, 2023-09-19:
Solve it for me:
`ln -s /usr/local/lib/python3.6/site-packages/cv2/python-3.6/cv2.cpython-36m-aarch64-linux-gnu.so cv2.so`