mdegans/nano_build_opencv #58: 3.2.0 fails to build

mdegans, 2021-05-27:
Hi mdegans,

First, I would like to thank you very much for providing this script. It allowed me to install opencv 4.4 flawlessly on my jetson nano jetpack 4.4, so really, thank you!

I'm trying to make an old piece of code to work on my jetson nano as well, and this project uses opencv 3.2.0.
I decided to try your script to install this opencv version but unfortunately it fails, after some time.
The fail seems to be related to cmake configuration, but as I have no knowledge of cmake, I'm struggling to find the reason why.

If by chance you have some free time and you would like to help me use your script to install opencv 3.2.0, [here](https://gist.github.com/lweingart/69d59ed422108f699daf85d8dc0a71e5) is a link to my terminal output, the CMakeOutput.log and the CMakeError.log

Thank you, happy new year!

Laurent

_Originally posted by @lweingart in https://github.com/mdegans/nano_build_opencv/issues/12#issuecomment-753536390_

mdegans, 2021-05-27:
@lweingart

I'm going through my issues and just saw your comment. I have no idea what's causing that. Can you provide system details so I can repro?

Will need:

* Jetpack Version
* Nano (A or B)
* Any third party customizations?

lweingart, 2021-05-27:
Hello @mdegans ,

Thank you very much for getting back to me.
Actually I did it differently in the end, so I should have closed this issue but completely forgot about it.
Also I don't remember on which device it was, so I'll be having difficulties to provide you with details.

Allow me to thank you once more for this script, it's really helpful, I'm using it on all my jetsons :-)

Have a great day

mdegans, 2021-05-27:
Glad you like the script! I'll close this then and if someone else has the issue, please re-open it.

Have a great day, AW!

closed 2021-05-27 by mdegans

bperiard, 2021-06-10:
Hi @mdegans , I have the same issue on my jetson xavier NX, with jetpack 4.5-b129.
I have been able to build opencv 3.4 and 4.4 flawlessly but I can't compile 3.2.
You can see my logs here
[CMakeOutput.log](https://github.com/mdegans/nano_build_opencv/files/6632712/CMakeOutput.log)
[CMakeError.log](https://github.com/mdegans/nano_build_opencv/files/6632713/CMakeError.log)

Thank you for your time!