mdegans/nano_build_opencv #63: I want to Install opencv dnn module to python3 but when i run script it automatically got installed to python2.7

8087482204, 2021-12-02:
[image]
I wanted opencv 4.1.1 to be installed on python3 but it got installed to python 2.7 how I can reinstall it to python3 or is there any command to make it use with python3.
Also there is preinstalled version of Opencv in python3 how i can remove it

MinhDung0803, 2021-12-29:
I have complied successfully with python3 by followed these steps:
1. Uninstall Python2 out of your Jetson
sudo apt purge -y python2.7-minimal

2. Replace default Python2 by Python3(and pip also)
sudo ln -s /usr/bin/python3 /usr/bin/python
sudo ln -s /usr/bin/pip3 /usr/bin/pip

3. Build opencv 4.4.0
sudo ./build_opencv.sh

Result:
[image]

I hope this help :)

sarvaatmak, 2022-03-16:
@MinhDung0803  @mdegans I did as you said and uninstalled python2 but then i run the script and observed that it again reinstalled python2 while ongoing process.
can you create separate scripts for python3 and python2 it might help a lot.
i am trying to install cv2 with your sctipt in jetpack 4.6 i think its installing version opencv 4.4 will update you the result i am installing on jetson xavier nx i.e if successful or not.
thanks.

mdegans, 2023-01-27:
The expected behavior is to build for both. I'll try to replicate when I get a chance.

@MinhDung0803 

These commands here...
```
sudo ln -s /usr/bin/python3 /usr/bin/python
sudo ln -s /usr/bin/pip3 /usr/bin/pip
```
...might break a lot of things on Linux for Tegra because `/usr/bin/python` is usually python2 and apt needs to know when a python version changes. Try instead:
```
sudo update-alternatives --config python
```

To see what versions are available and switch. It will still likely break a lot of things, but hopefully fewer. Recent Ubuntu versions (and likely L4T) use `python` instead of `python3`, but only if you install a `python-is-python3` package for it.

```
$ python
Command 'python' not found, did you mean:
  command 'python3' from deb python3
  command 'python' from deb python-is-python3
```