mdegans/nano_build_opencv #65: Build OpenCV with CUDA on Jetson AGX Xavier

habjoel, 2022-05-18:
Hi @mdegans 

I would like to use your script to build OpenCV with CUDa support on the Jetson AGX Xavier. 
There is already a OpenCV installed on the device through Jetpack (but without cuda). Do I need to uninstall it before running the script (sudo apt-get purge -y '*opencv*') or does it work without that?

Furthermore, is there any reason why this script should not work for OpenCV version > 4.4.0?

mdegans, 2022-08-03:
> Hi @mdegans
> 
> I would like to use your script to build OpenCV with CUDA support on the Jetson AGX Xavier. There is already a OpenCV installed on the device through Jetpack (but without cuda). Do I need uninstall it before running the script (sudo apt-get purge -y '*opencv*') or does it work without that?
> 
> Furthermore, is there any reason why this script should not work for OpenCV version > 4.4.0?

It's not mandatory to remove a preexisting version of OpenCV so long as you know your build system and/or python paths. Some Nvidia software requires 4.1.1 anyway, I think, and as such apt might remove more than you'd expect (because it would break that software anyway).

mdegans, 2022-08-03:
> @mdegans The script worked flawlessly for OpenCV 4.5.0. :) The only thing I am confused about is that there is no /usr/include/opencv4 or /usr/lib/python3.6/dist-packages/cv2 directory, which exist for the native OpenCV installation. Why is that and should I be worried about it?
> 
> Furthermore, I am planning to use OpenCV together with ROS Melodic and I am hoping that ROS will use my newly installed OpenCV with CUDA... do you have any experience with that?

You'll have to look into OpenCV's cmake to see how it figures out the python path (and the way that worked it [changed recently](https://github.com/opencv/opencv/commit/d6891c705e269465222049510ef0920a47507ca9#diff-d55511776794dc61d4962ae7a9025c0fa54a10b3db01de0079c6ad031f259851R180)). If you run the script from within a venv, it'll probably pick the venv's location to install the python module. In any case you *should* be able to copy and paste the `cv2` directory wherever you want it without breaking anything.

HXB-1997, 2023-11-08:
what's your jetpack version? @habjoel  my jetpack is 5.1.2, when I run the bash, terminal says multipul dependency unmet,and the dependency is not to be installed