mdegans/nano_build_opencv #67: Building OpenCV for Orin System

randyg3000, 2022-06-03:
To build OpenCV 4..5.5 (latest) for Orin I had to add 8.7 to "CUDA_ARCH_BIN"

-D CUDA_ARCH_BIN=5.3,6.2,7.2,8.7

deviceQuery showed:

Device 0: "Orin"
  CUDA Driver Version / Runtime Version          11.4 / 11.4
  CUDA Capability Major/Minor version number:    8.7

Also, because of a quirk with OpenCV 4.5.5 I had to specify

-D PYTHON3_PACKAGES_PATH=/usr/lib/python3/dist-packages

(See [https://github.com/opencv/opencv/issues/21359])

Finally, I commented out "v4l2ucp" which doesn't seem to be supported anymore.

mdegans, 2022-06-08:
Thanks for the update and the heads-up on that issue. Can you submit a PR with the `CUDA_ARCH_BIN` fix? I do not have an Orin to test on, but I will check it doesn't break AGX, NX and Nano (can't imagine how, but I will test it).

Less sure about the Python path workaround since it's breakage on OpenCV's side. A hardcoded path might break expectations of people who have been relying on some other `python3` being [detected automatically](https://github.com/opencv/opencv/commit/d6891c705e269465222049510ef0920a47507ca9#diff-d55511776794dc61d4962ae7a9025c0fa54a10b3db01de0079c6ad031f259851R180).

Perhaps it's more useful to just fail fast for version "4.5.5" and print a message to use another version.

HarrySm, 2022-07-27:
Hello, I installed opecnCV from the shell script and modified it as @randyg3000 said. However, I can't use it, every time I do <code>
#include <opencv2/opencv.hpp>
</code>
I receive an error:
fatal error: opencv2/opencv.hpp: No such file or directory
   41 | #include <opencv2/opencv.hpp>

nakai-omer, 2022-09-07:
@mdegans I have created a PR for removing v4l2ucp and adding 8.7 for CUDA_ARCH_BIN

PR #69 (2023-01-27): Add support Orin and more up to date ubuntu
Fixed #67 & #66
closed 2023-01-27 by mdegans