mdegans/nano_build_opencv #71: Logging is preventing multiple jobs

arpaterson, 2022-10-04:
(On a Jetson Orin)

This script does not use more than one core, regardless of the value in `$CPUS`.

I have been having issues building opencv and getting it working with realsense-ros, so I've been reconfiguring and building using my own script, and the one in this git. I noticed mine is dramatically faster, despite being almost identical aside from a few flags.

Changing nano_build_opencv's make command 
`make install 2>&1 | tee -a install.log` 
to 
`make -j$(($(nproc)-1))` ,
removes the redirection of the output to the log file but gives a dramatic speed up, with jtop showing multiple or all cores pegged at near 100%.

Suspect this affects all systems.

mdegans, 2023-01-27:
Thanks for letting me know. I haven't tested this recently, but I can see why multiple threads might be blocked writing to the same log file. There might be some options to `make` or `tee` that can resolve this.

Possibly `ninja` build could do a better job, although I don't know if `cmake -G ninja` works with OpenCV yet. Thanks for filing the issue. I'll look into this if I get a chance. Otherwise, feel free to submit a PR disabling logging for the moment.

arpaterson, 2023-01-27:
No worries!