mdegans/nano_build_opencv #73: In jtop still cuda support No

mertmirac13, 2023-03-27:
I am a newbie and im using jetson nano for the first time and my problem is after compiling and installing cuda in jtop still cuda support says No. Is there a thing i should do after installation?

lweingart, 2023-04-16:
Hi, I guess my issue is somewhat related to your.
I have a Jetson AGX Orin that has opencv 4.5.4 installed by default, without cuda support.
I used this repo on my other Jetson devices and it worked perfectly fine.
This time, the build and compilation went well, the script was executed without errors.

However, when I import cv2, it's still the old version that is used, I don't know why.
Additionaly, I cannot find a new cv2 directory (used to be created in /usr/local/lib)
I find a lot of libopencv* lib files there, created by this install script, but I don't know how to use the newly installed opencv version.

Anyone could help ?
Thank you very much in advance

Cheers

lweingart, 2023-04-16:
Hi again,

I was actually able to solve my issue by applying the modifications suggested here: https://github.com/mdegans/nano_build_opencv/issues/67
As a summary:
add 8.6 to the -D CUDA_ARCH_BIN option
modify -D CUDNN_VERSION with 8.6 value
and add this option: -D PYTHON3_PACKAGES_PATH=/usr/lib/python3/dist-packages

The last option, however, will have this newly installed opencv overwrite the default one, which in my case was just perfectly fine with me.
Bye

mertmirac13, 2023-04-16:
Hi,  my issue was same as yours but i solved it flashing my jetson nano and
build cuda again. Thanks for your answers.

şunu yazdı: