mdegans/nano_build_opencv #77: jetson nano errors to install it

tasosman, 2023-07-07:
hi i try as in the video but without goodd results something say for unsupported gpu
 see what is writing:
[ 24%] Building NVCC (Device) object modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o
nvcc fatal   : Unsupported gpu architecture 'compute_87'
CMake Error at cuda_compile_1_generated_gpu_mat.cu.o.RELEASE.cmake:221 (message):
  Error generating
  /tmp/build_opencv/opencv/build/modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_gpu_mat.cu.o
modules/core/CMakeFiles/opencv_core.dir/build.make:63: recipe for target 'modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o' failed
make[2]: *** [modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o] Error 1
CMakeFiles/Makefile2:2587: recipe for target 'modules/core/CMakeFiles/opencv_core.dir/all' failed
make[1]: *** [modules/core/CMakeFiles/opencv_core.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
[sudo] password for nano: 
[  1%] Built target ade
[  1%] Built target gen-pkgconfig
[  1%] Built target opencv_highgui_plugins
[  8%] Built target libwebp
[  9%] Built target libopenjp2
[ 15%] Built target IlmImf
[ 19%] Built target libprotobuf
[ 19%] Built target quirc
[ 22%] Built target carotene_objs
[ 23%] Built target tegra_hal
[ 23%] Built target carotene
[ 24%] Built target ittnotify
[ 24%] Built target opencv_videoio_plugins
[ 24%] Built target opencv_cudev
[ 24%] Building NVCC (Device) object modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o
nvcc fatal   : Unsupported gpu architecture 'compute_87'
CMake Error at cuda_compile_1_generated_gpu_mat.cu.o.RELEASE.cmake:221 (message):
  Error generating
  /tmp/build_opencv/opencv/build/modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/./cuda_compile_1_generated_gpu_mat.cu.o

modules/core/CMakeFiles/opencv_core.dir/build.make:63: recipe for target 'modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o' failed
make[2]: *** [modules/core/CMakeFiles/cuda_compile_1.dir/src/cuda/cuda_compile_1_generated_gpu_mat.cu.o] Error 1
CMakeFiles/Makefile2:2587: recipe for target 'modules/core/CMakeFiles/opencv_core.dir/all' failed
make[1]: *** [modules/core/CMakeFiles/opencv_core.dir/all] Error 2
Makefile:162: recipe for target 'all' failed
make: *** [all] Error 2
Do you wish to remove temporary build files in /tmp/build_opencv ? 
(Doing so may make running tests on the build later impossible)
Y/N Y
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/23': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/19': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/1': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/100': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/3': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/7': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/9': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/10': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/8': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/18': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/count.txt': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/13': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/20': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/21': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/16': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/4': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/15': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/6': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/14': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/2': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/5': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/17': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/12': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/11': Permission denied
rm: cannot remove '/tmp/build_opencv/opencv/build/CMakeFiles/Progress/22': Permission denied

jpuum, 2023-08-07:
I assume you have not changed the default CMake flags if you face this error. The problem causing `nvcc fatal : Unsupported gpu architecture 'compute_87'` stems from `CUDA_ARCH_BIN=5.3,6.2,7.2,8.7` flag. Try removing 8.7 and run it again.

tasosman, 2023-08-12:
Thank you is working fine now

hanifazhar, 2024-03-30:
> I assume you have not changed the default CMake flags if you face this error. The problem causing `nvcc fatal : Unsupported gpu architecture 'compute_87'` stems from `CUDA_ARCH_BIN=5.3,6.2,7.2,8.7` flag. Try removing 8.7 and run it again.

how to remove it

eminberkayd, 2024-05-22:
> > I assume you have not changed the default CMake flags if you face this error. The problem causing `nvcc fatal : Unsupported gpu architecture 'compute_87'` stems from `CUDA_ARCH_BIN=5.3,6.2,7.2,8.7` flag. Try removing 8.7 and run it again.
> 
> how to remove it

just remove the `,8.7` part. That means the line should be `CUDA_ARCH_BIN=5.3,6.2,7.2`

tasosman, 2024-05-22:
thank you i had fixxed that with what are you wrote