mdegans/nano_build_opencv #78: Can not install it in docker

mmldxxn, 2023-07-12:
Hi,
I tried to use this command install opencv in docker, but it seems there is an error.
This is part of my Dockerfile
I put the build_opencv.sh under assets/${ARCH} 
-----------
#! install ML Related Stuff
COPY assets/${ARCH} "${REPO_PATH}/install"
RUN "${REPO_PATH}/install/install.sh"
RUN "${REPO_PATH}/install/build_opencv.sh"

----
this is output.

The command '/bin/bash -c "${REPO_PATH}/install/build_opencv.sh"' returned a non-zero code: 1
An error occurred while building the project image.

could you please help me with that?

kangaroo-0000, 2023-08-02:
Same issue. Following

kangaroo-0000, 2023-08-02:
Same issue. following. [bash.txt](https://github.com/mdegans/nano_build_opencv/files/12236380/bash.txt)